AndrewEllis93/Pressure_Linear_Advance_Tool #7: start/end macros don't match klipper default names or parameters

strayr, 2022-12-10:
Klipper [sample-macros.cfg](https://github.com/Klipper3d/klipper/blob/master/config/sample-macros.cfg) uses

```
[gcode_macro START_PRINT]
gcode:
    {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
    {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(190)|float %}

# ...

[gcode_macro END_PRINT]
```

You are using different macro names and parameters, presumanbly becasue voron broke with convention?

AndrewEllis93, 2022-12-14:
Yep. 
Everyone has completely different setups, it's nearly impossible to write a tool for Klipper because of that.
Most I've seen use PRINT_START / PRINT_END, but of course my experience is lopsided from spending most of my time in the Voron Discord.

AndrewEllis93, 2022-12-31:
I added START_PRINT as an alternate (commented out)

closed 2022-12-31 by AndrewEllis93