AndrewEllis93/Pressure_Linear_Advance_Tool #11: Selectable Z zero location

hoverdj, 2022-12-15:
My printer is a FalshForge Creator Pro (MakerBot replica, basically) that has the bed rising and lowering for the Z axis. Zero is at top.

When running your code, it appears it assumes Z0 is at the bottom, as is the case with most printers (I think). 

It would be nice if your tool could allow for that as an option. Maybe it does and I'm just not seeing it?

Thanks for your consideration!
D-

AndrewEllis93, 2022-12-15:
Currently it does not have that - but I don't think it would be too difficult to add, pretty much just invert all the Z coordinates, right?

hoverdj, 2022-12-15:
Yes, that is the way I understand it. ThanksD-

hoverdj, 2022-12-15:
Never mind. It magically started working. No idea what I did wrong last time. 
thanks for being willing to do it though. 
D-

closed 2022-12-15 by hoverdj