AndrewEllis93/Pressure_Linear_Advance_Tool #19: Can't set End K value higher than 0.2

2probro, 2023-02-17:
**Describe the bug**
I want to use a Linear advance K-value of 1 or higher on my bowden ender 3. But set "End K value" to 1, it doesnt work anymore. The "End K value" gets highlighted in yellow and I can't download my gcode. Why is this not possible, k values for bowden setup are limited by the page to 0.2, which is too low for my printer.

**Klipper Error Message (if applicable)**

**G-code File**
Not able to set my end k value, so couldn't export my gcode

**Screenshots**
[image]
[image]

2probro, 2023-02-18:
The whole section for pressure advance max 0.2 should be removed:

`    if (!validationFail) {
      // only check if above checks pass
      // Check if PA smooth exceeds 0.2
      if (config.pa_smooth && config.pa_end > 2){
          $("label[for=PA_END]").addClass("invalid");
          $("#warning1").text(
            "PA smooth cannot exceed 0.2."
          );
          $("#warning1").addClass("invalid");
          $("#warning1").show();
          invalid = 1;
          validationFail = true;
      }
    }`

2probro, 2023-02-18:
I also noticed that it says "Pa smooth" while using marlin's linear advance. This is also wrong right?

kotarokun, 2023-06-18:
**A workaround** for me is after setting End-K-Value to 1 change Firmware to Klipper and change back to Marlin 1.1.9+
it works and i can download my gcode.
maybe there is an idea for correction?

LittleNyanCat, 2023-07-24:
Also having this issue, particularly annoying when the dialog box specifically recommends starting out with 1 if you have a bowden...

nesb0t, 2023-07-28:
This bug was introduced [here](https://github.com/AndrewEllis93/Pressure_Linear_Advance_Tool/commit/6ac1f1c9cc3a63111ef17c48f64d90fe981fc1e1) to fix #14 so simply removing the validation check isn't the right fix. 

The cause was that `config.pa_smooth` is returning `true` even when you didn't turn it on. I was working on troubleshooting it so I could submit a PR but for some reason I can no longer replicate it and things are now working as expected. Ha. I have a feeling it's related to filling out the form in a particular order and something is turning that option on, but after running in to this issue repeatedly, I can't replicate it anymore so I have no idea. 

If you find yourself running in to this issue and don't want to modify the js as mentioned above, an easy fix is to enable the `Advanced Mode` checkbox at the top and then enable and then disable `Increment Smooth Time Instead` under `Pressure Advance Stepping`. This will force the variable to `false` and it should generate.

LittleNyanCat, 2023-08-01:
That validation should only be checked if the selected firmware is Klipper, as #14 is a Klipper-specific issue.

I'm having issues with `Marlin 1.1.9+`/`Marlin 1.1.8 & Older` selected, where there isn't a `Increment Smooth Time Instead` option which makes things harder. 

I've resorted to setting it to Klipper, doing your workaround, messing with some of the values, switching back to Marlin and then *sometimes* (haven't figured out exactly what makes it happen) it starts letting me use any max PA value with Marlin selected.

msticninja, 2023-08-07:
Keep Firmware type as Klipper, but enable Advanced mode with the checkbox at the top of Settings, and Uncheck "Increment Smooth Time Instead" in the PA section. Then you can use the full range of End Values without changing the firmware type. Once you find your PA, you can recheck the "Increment Smooth Time Instead" box and tune your Firmware's PA Smoothing Time.

AndrewEllis93, 2024-01-11:
I set that validation check to only happen if Klipper is selected. Can you please see if this issue is still occurring?