AndrewEllis93/Pressure_Linear_Advance_Tool #20: Add an options to enable Label Objects

lucascarvalho, 2023-03-29:
**Is your feature request related to a problem? Please describe.**
Add an option to enable Label Objects.

lucascarvalho, 2023-06-12:
KAMP

RodentVienna, 2023-09-26:
HI, i run into the same issue. I also use KAMP, and therefore cannot proceed with the GCODE. KAMP expects EXCLUDE_OBJECT_DEFINE, which is not part of your gcode. i am currently looking for temporarily disabling KAMP, but it seems KAMP is deeply integrated once installed :(

kenjesse, 2023-10-08:
Also using KAMP (Klipper Adaptive Bed Meshing)  For me I'm able to run the print after editing the start code in the gcode file but without the Object Label instead of only probing the area of the bed where the print will be is does the entire bed 8x8, takes a little while.  Love and recommend your Tuning Guide to anyone that will listen to me about 3D Printing.  Thanks for all your efforts Sir.

ErockSaint, 2023-10-19:
This would be great. I'm using the adaptive mesh as well as purge. current issue is that once it does the whole mesh, it goes to the adaptive purge, then gives and out of range error because it's trying to put that purge line out in thin air. Like mentioned above, I think KAMP id dug in too deep to try and disable it for one print.

Not sure the object label will help in my specific case, but would be nice if it did.