AndrewEllis93/Pressure_Linear_Advance_Tool #29: Add object labeling (support KAMP)

ronpandolfi, 2023-10-08:
**Is your feature request related to a problem? Please describe.**
Because the tuning model is not labeled, KAMP's adaptive meshing resorts to probing across the whole bed 

**Describe the solution you'd like**
Add object labels to the generated stl.

remizca, 2023-11-04:
i'm having the same issue. have you solved this yet?

ronpandolfi, 2023-11-04:
No, I was hoping someone else would pick up this issue.

remizca, 2023-11-04:
oh okay. i was worried at first because i thought something was wrong with my KAMP setup and then i tried printing a square and KAMP works just fine

Squid116, 2023-12-23:
I have added an initial implementation for exclude objects for KAMP.  Printing now, seems to be working as desired.

https://github.com/AndrewEllis93/Pressure_Linear_Advance_Tool/pull/34

LAP87, 2024-03-24:
Any update to this? my printer smacks the toolhead quite violently into the far corner to print purge after wasting a bunch of time probing the full 350 bed..