AndrewEllis93/Pressure_Linear_Advance_Tool #31: Ehm, dangerous tool... It does not accept z_offset

tomdelahaba, 2023-11-11:
**Describe the bug**
Well, I thought I made a mistake by myself... But after 2 tries, this tool basically changed my Voron 2.4 into CNC 2.4... So, first of all. I was able to use "without a problem" this calibration tool with TAP. Unfortunately, because of the weight, x position, vibrations, speeds, I moved back to "original" inductive probe (Inductive probe is working fine). I followed again the guideline from scratch (first layer was calibrated by "paper" and later on I went through the first layer squish, I set the correct z_offset). Printing something is without any problem. Unfortunately, when I used the calibration tool (pattern method) to calibrate the pressure advance, after generating the gcode and printing it, the nozzle simply scratched the printing plate... I was soooo f***ed up on myself, but I thought I made some mistake with the settings. I set everything from scratch again and the same... That nozzle is printing like 0.5-1mm bellow where it should.

**Klipper Error Message (if applicable)**
N/A

**G-code File**
[pa_pattern.gcode.gz](https://github.com/AndrewEllis93/Pressure_Linear_Advance_Tool/files/13323955/pa_pattern.gcode.gz)

**Screenshots**
Well, the only screenshot of this CNC I can provide is how the printing plate is scratched and you will laugh to me... :( So nice plate, it was...

AndrewEllis93, 2024-01-11:
It looks like your PRINT_START macro is commented out in the gcode. Can you show what you pasted in the start g-code box?

HackThePlanetKC, 2025-05-21:
[pa_pattern (5).txt](https://github.com/user-attachments/files/20370168/pa_pattern.5.txt)

I'm getting the same problem. My z-offset is set in Orca to 1.4mm, and I've made sure it was in my printer.cfg. Can't seem to get the pa patterns to accept it.